SodaqMoja/SodaqOne-UniversalTracker #1: Questions related to battery + board temperature

brocaar, 2016-07-31:
Hi! I've received the SodaqOne yesterday and with the pre-installed tracker and boot menu, it was really easy to get it up and running 👍 

Most values I've been able to parse, but some of them I'm not sure if I understand them correctly:

**Battery**:

In the readme the battery voltage is documented as the value beween 3-4.5V. Does this mean the value defines the steps between 3 and 3.5V? So to get the voltage, I have to make the following calculation?

`3.0 + (1.5/255 * value) = voltage`

**Board temperature**:
Given that I'm getting values > 200 back, and my board is not on fire 😉 , how should this be interpreted? Should I divide the result by 10?

---

```
2016/07/31 10:16:03 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-51 Data:[195 179 157 87 89 229 60 64 56 31 190 47 229 2 5 0 1 0 100 5 3]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 10:16:03 +0200 CEST BatteryVoltage:3.523529411764706 BoardTemperature:229 Lat:52.3780156 Long:4.8574398 Altitude:5 Speed:1 Course:100 NumSatellites:5 TimeToFix:3}
```

brocaar, 2016-07-31:
Some more board temperature readings. Interesting thing is that it went down for a while (to around 10), but when I moved it, it went up to > 200 again.

```
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:38:26 +0200 CEST BatteryVoltage:3.5058823529411764 BoardTemperature:9 Lat:52.3780368 Long:4.8569793 Altitude:65535 Speed:0 Course:0 NumSatellites:5 TimeToFix:26}
2016/07/31 11:39:37 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-60 Data:[89 199 157 87 85 12 194 66 56 31 27 30 229 2 255 255 1 0 39 5 37]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:39:37 +0200 CEST BatteryVoltage:3.5 BoardTemperature:12 Lat:52.3780802 Long:4.856988299999999 Altitude:65535 Speed:1 Course:39 NumSatellites:5 TimeToFix:37}
2016/07/31 11:40:06 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-59 Data:[118 199 157 87 86 6 27 66 56 31 149 71 229 2 57 0 0 0 127 5 6]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:40:06 +0200 CEST BatteryVoltage:3.5058823529411764 BoardTemperature:6 Lat:52.378063499999996 Long:4.8580501 Altitude:57 Speed:0 Course:127 NumSatellites:5 TimeToFix:6}
```

After moving it a bit (without restart):

```
2016/07/31 11:41:03 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-64 Data:[175 199 157 87 86 254 85 64 56 31 38 57 229 2 39 0 1 0 61 5 3]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:41:03 +0200 CEST BatteryVoltage:3.5058823529411764 BoardTemperature:254 Lat:52.3780181 Long:4.8576806 Altitude:39 Speed:1 Course:61 NumSatellites:5 TimeToFix:3}
2016/07/31 11:42:07 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-59 Data:[239 199 157 87 85 247 96 74 56 31 230 36 229 2 34 0 2 0 156 4 7]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:42:07 +0200 CEST BatteryVoltage:3.5 BoardTemperature:247 Lat:52.3782752 Long:4.857162199999999 Altitude:34 Speed:2 Course:156 NumSatellites:4 TimeToFix:7}
```

brocaar, 2016-07-31:
And leaving it on the same place for a while (going from 240 down to 7):

```
2016/07/31 11:46:09 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-64 Data:[225 200 157 87 87 240 242 71 56 31 255 26 229 2 255 255 4 0 0 4 9]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:46:09 +0200 CEST BatteryVoltage:3.511764705882353 BoardTemperature:240 Lat:52.378212999999995 Long:4.8569087 Altitude:65535 Speed:4 Course:0 NumSatellites:4 TimeToFix:9}
2016/07/31 11:47:12 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-61 Data:[32 201 157 87 85 244 64 54 56 31 231 31 229 2 47 0 1 0 254 5 12]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:47:12 +0200 CEST BatteryVoltage:3.5 BoardTemperature:244 Lat:52.377759999999995 Long:4.8570343 Altitude:47 Speed:1 Course:254 NumSatellites:5 TimeToFix:12}
2016/07/31 11:48:24 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-63 Data:[104 201 157 87 86 251 92 53 56 31 206 24 229 2 51 0 11 0 34 5 24]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:48:24 +0200 CEST BatteryVoltage:3.5058823529411764 BoardTemperature:251 Lat:52.3777372 Long:4.8568526 Altitude:51 Speed:11 Course:34 NumSatellites:5 TimeToFix:24}
2016/07/31 11:49:06 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-63 Data:[146 201 157 87 86 251 99 83 56 31 121 24 229 2 255 255 0 0 0 5 6]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:49:06 +0200 CEST BatteryVoltage:3.5058823529411764 BoardTemperature:251 Lat:52.3785059 Long:4.8568441 Altitude:65535 Speed:0 Course:0 NumSatellites:5 TimeToFix:6}
2016/07/31 11:50:09 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-63 Data:[209 201 157 87 86 251 30 79 56 31 165 28 229 2 255 255 2 0 0 4 9]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:50:09 +0200 CEST BatteryVoltage:3.5058823529411764 BoardTemperature:251 Lat:52.378396599999995 Long:4.8569509 Altitude:65535 Speed:2 Course:0 NumSatellites:4 TimeToFix:9}
2016/07/31 11:51:15 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-63 Data:[19 202 157 87 86 254 195 70 56 31 207 26 229 2 255 255 8 0 183 4 15]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:51:15 +0200 CEST BatteryVoltage:3.5058823529411764 BoardTemperature:254 Lat:52.378182699999996 Long:4.8569039 Altitude:65535 Speed:8 Course:183 NumSatellites:4 TimeToFix:15}
2016/07/31 11:52:13 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-63 Data:[77 202 157 87 85 1 129 64 56 31 99 31 229 2 35 0 1 0 0 4 13]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:52:13 +0200 CEST BatteryVoltage:3.5 BoardTemperature:1 Lat:52.3780225 Long:4.8570211 Altitude:35 Speed:1 Course:0 NumSatellites:4 TimeToFix:13}
2016/07/31 11:53:32 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-63 Data:[156 202 157 87 85 4 202 61 56 31 159 33 229 2 11 0 1 0 69 4 32]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:53:32 +0200 CEST BatteryVoltage:3.5 BoardTemperature:4 Lat:52.377953 Long:4.8570782999999995 Altitude:11 Speed:1 Course:69 NumSatellites:4 TimeToFix:32}
2016/07/31 11:54:06 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-63 Data:[190 202 157 87 86 4 87 46 56 31 216 2 229 2 37 0 1 0 0 5 6]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:54:06 +0200 CEST BatteryVoltage:3.5058823529411764 BoardTemperature:4 Lat:52.377557499999995 Long:4.8562904 Altitude:37 Speed:1 Course:0 NumSatellites:5 TimeToFix:6}
2016/07/31 11:55:08 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-60 Data:[252 202 157 87 86 4 192 53 56 31 4 19 229 2 21 0 1 0 0 4 8]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:55:08 +0200 CEST BatteryVoltage:3.5058823529411764 BoardTemperature:4 Lat:52.377747199999995 Long:4.8567044 Altitude:21 Speed:1 Course:0 NumSatellites:4 TimeToFix:8}
2016/07/31 11:56:10 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-63 Data:[58 203 157 87 86 5 34 67 56 31 84 37 229 2 7 0 1 0 0 5 10]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:56:10 +0200 CEST BatteryVoltage:3.5058823529411764 BoardTemperature:5 Lat:52.3780898 Long:4.8571732 Altitude:7 Speed:1 Course:0 NumSatellites:5 TimeToFix:10}
2016/07/31 11:57:17 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-56 Data:[125 203 157 87 86 6 46 83 56 31 118 31 229 2 255 255 4 0 0 5 17]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 11:57:17 +0200 CEST BatteryVoltage:3.5058823529411764 BoardTemperature:6 Lat:52.378500599999995 Long:4.857023 Altitude:65535 Speed:4 Course:0 NumSatellites:5 TimeToFix:17}
[... cut ...]
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 12:03:05 +0200 CEST BatteryVoltage:3.5 BoardTemperature:7 Lat:52.378006899999995 Long:4.8568919 Altitude:22 Speed:0 Course:0 NumSatellites:5 TimeToFix:5}
2016/07/31 12:04:06 topic: application/0202020202020202/node/0101010101010101/rx, payload: {DevEUI:0101010101010101 FPort:1 GatewayCount:1 RSSI:-67 Data:[22 205 157 87 85 7 167 56 56 31 14 20 229 2 43 0 0 0 73 5 6]}
decoded record: {DevEUI:0101010101010101 Time:2016-07-31 12:04:06 +0200 CEST BatteryVoltage:3.5 BoardTemperature:7 Lat:52.377821499999996 Long:4.856731 Altitude:43 Speed:0 Course:73 NumSatellites:5 TimeToFix:6}
```

trooster, 2016-07-31:
I asked the same question in the forum:
http://forum.sodaq.com/t/interpreting-payload-data-from-the-sodaqone-universaltracker/374

Battery voltage is: (3000+10*<data>)/1000 
If I interpret the source code

KKjelstad, 2016-08-10:
It looks like there is an issue with the "getBoardTemperature" function. It reads two bytes from the LSM303, but it returns only 1 byte.

```
/**
 * Returns the board temperature.
*/
int8_t getBoardTemperature()
{
    setLsm303Active(true);

    uint8_t tempL = lsm303.readReg(LSM303::TEMP_OUT_L);
    uint8_t tempH = lsm303.readReg(LSM303::TEMP_OUT_H);

    int16_t temp = (int16_t)(((uint16_t)tempH << 8) | tempL);

    setLsm303Active(false);

    return temp;
}
```

The datasheet states 

> Temperature data (8 LSB/deg - 12-bit resolution). The value is expressed as
two’s complement.

If the return statement of the getBoardTemperature() is changed to this:

`return (temp + 4) / 8`

Then I think the value should be a correct celsius degree

Unless I have misunderstood something here?

keestux, 2016-08-10:
@KKjelstad without looking at the datasheet I can tell that the code looks suspicious.

The LSM303D datasheet states:
> Temperature sensor data. Temperature data is stored as two’s complement data in 12-bit 
format, right-justified.

To me that means we need to do:

    int16_t temp = (int16_t)(((uint16_t)tempH << 8) | tempL);
    temp <<= 4;

And make the function type int16_t, or else do

    int16_t temp = (int16_t)(((uint16_t)tempH << 8) | tempL);
    temp <<= 4;
    return temp >> 8;

Can someone test this?

KKjelstad, 2016-08-11:
@keestux I have done some testing by reading out the H and L bytes. First I cooled down the board.

| H | L | Decimal | div by 8 | offset by 20 |
| ------------- | ------------- | ------------- | ------------- | ------------- |
| FF | A9 | -87 | -10.875 | 9.125 |
| FF | B9 | -71 | -8.875 | 11.125 |
| FF | D0 | -48 | -6 | 14 |
| FF | E1 | -31 | -3.875 | 16.125 |
| FF | EE | -18 | -2.25 | 17.75 |
| FF | F8 | -8 | -1 | 19 |
| 00 | 00 | 0 | 0 | 20 |
| 00 | 07 | 7 | 0.875 | 20.875 |
| 00 | 0B | 11 | 1.375 | 21.375 |
| 00 | 0F | 15 | 1.875 | 21.875 |
| 00 | 13 | 19 | 2.375 | 22.375 |
| 00 | 15 | 21 | 2.625 | 22.625 |
| 00 | 15 | 21 | 2.625 | 22.625 |
| 00 | 18 | 24 | 3 | 23 |
| 00 | 18 | 24 | 3 | 23 |
| 00 | 1B | 27 | 3.375 | 23.375 |
| 00 | 1B | 27 | 3.375 | 23.375 |
| 00 | 19 | 25 | 3.125 | 23.125 |
| 00 | 1C | 28 | 3.5 | 23.5 |
| 00 | 1C | 28 | 3.5 | 23.5 |
| 00 | 28 | 40 | 5 | 25 (applied heat source) |
| 00 | 40 | 64 | 8 | 28 |
| 00 | 4E | 78 | 9.75 | 29.75 |
| 00 | 59 | 89 | 11.125 | 31.125 |
| 00 | 5E | 94 | 11.75 | 31.75 |
| 00 | 50 | 80 | 10 | 30|

This is the code I used:

```
setLsm303Active(true);

uint8_t tempL = lsm303.readReg(LSM303::TEMP_OUT_L);
uint8_t tempH = lsm303.readReg(LSM303::TEMP_OUT_H);

int16_t temp = (int16_t)(((uint16_t)tempH << 8) | tempL);

#ifdef DEBUG
DEBUG_STREAM.print("Temp: ");
DEBUG_STREAM.print(tempH, HEX);
DEBUG_STREAM.print(" ");
DEBUG_STREAM.print(tempL, HEX);
DEBUG_STREAM.print(" ");
DEBUG_STREAM.print(temp, DEC);
#endif

setLsm303Active(false);

```

By using 20 + ((temp + 4) / 8) it seems to fit with Celsius, but it is hard to interpret the datasheet to this.

alextsam, 2016-08-11:
@keestux @KKjelstad The original code should be fine.
Please see the suggestion from GabrielNotman at the forum thread (a couple of posts above here, http://forum.sodaq.com/t/interpreting-payload-data-from-the-sodaqone-universaltracker/374). The suggestion has to do with the sampling rate of the sensor.

@KKjelstad there is no problem with the conversion from 16-bit int to 8-bit int because the operating temperature range is -40..+85 (so you will/should never see a value outside this range), which fits well in a signed byte.

keestux, 2016-08-11:
@alextsam Hmm, to me this makes no sense. First, I don't see any match with the description in the datasheet. Second, why would you want to cripple a function to 8 bit if the device gives 12 bit?

@KKjelstad the datasheet is not clear at all. I just found some other software on github (PX4 Firmware) which states this in the source code, lsm303d.cpp:

    /* remember the temperature. The datasheet isn't clear, but it
     * seems to be a signed offset from 25 degrees C in units of 0.125C
    */

Then some other software (dcms_media, LSM303D_spi.ino) has:

    TempRaw    = (Buffer_TEMP[1] << 8) | Buffer_TEMP[0];
    // Temperature (8LSB/C)
    Temp   = 25 + (float)TempRaw / 8;

This is similar to what you are experiencing.

alextsam, 2016-08-11:
@keestux Could you elaborate what doesn't match with the datasheet?

The value is not "crippled" as long as it is Celsius in the range -40..+85. Plus the Lora packet specs allowed for only 1 byte, in Celsius.
Having said that, I see now that to have Celsius, a division by 8 is needed (8 LSB/C). Which then makes me wonder how for some there were sensible values returned.

Regarding that the temperature given is an offset, unfortunately I cannot find anything related in the datasheet.

alextsam, 2016-08-11:
@KKjelstad Could you try to replace the "return temp;" with "return temp / 8 + 25;"

KKjelstad, 2016-08-11:
@alextsam @keestux I have replaced the return statement with:

`return 20 + ((temp + 4) / 8);`

Not sure how much heat the device generates it self, but when idling in the office here the temp value stabilized on 28 which results in 3.5. Adding an offset of 25 makes it too high compared to the room temperature so I am currently testing 20 which seems to result in a better match.

Adding 4 to the temp value is for correct rounding.

keestux, 2016-08-11:
@alextsam The datasheet states:

> Temperature data is stored as two’s complement data in 12-bit format, right-justified.

I don't see that coming back in the logic. The datasheet should really explain what that means, but I think the upper 4 bits could be zero. If that is the case you first need to shift left, and back to do sign extension. No?

Furthermore, is there a need to drop the accuracy? Why not make the function return a float?

mrjdomingus, 2016-08-12:
Hi, @alextsam @KKjelstad , I have also been toying with the SODAQ One temperature sensor and can confirm the readings of @KKjelstad i.e.
- ambient temperature below 20 degrees Celsius will result in a negative (two's complement) value for the TEMP_OUT_H | TEMP_OUT_L combo.
- ambient temperature above 20 degrees Celsius will result in a positive (two's complement) value for the TEMP_OUT_H | TEMP_OUT_L combo.
I have experimented with ambient temperatures ranging from 15 degrees C tot 70 degrees C. Power for the SODAQ ONE was supplied by USB. 

The raw reading appears to increase by a value of 8 per degree Celsius temperature change, so division by 8 IS necessary to calculate a usable temperature value while factoring in the 20 degree C offset. I have used the following formula:

    int16_t temp = (int16_t)(((uint16_t)tempH << 8) | tempL);
    float temperature = 20.0 + (float)temp / 8.0;

@KKjelstad I do not understand the need to add 4 to the intermediate result. Can you elaborate? 

Division by 8 will also make the final result fit in a (signed) byte for the operating range of -40 to 85 degrees C. It is doubtful if returning a float instead of an int would improve 'accuracy' given the still fuzziness of the calculation of the final result.

I empathize with the frustration of @keestux as the datasheet is partly lacking or ambiguous with regard to relevant details. A temperature offset of 20 degrees C is mentioned nowhere, a temperature value of 25 degrees C is mentioned in several places in the datasheet, but is unclear if this is to be interpreted as an offset for calculation or as a baseline for sensor characteristics.

BTW, I am referencing the LSM303D datasheet as found here: [http://www.st.com/resource/en/datasheet/lsm303d.pdf](http://www.st.com/resource/en/datasheet/lsm303d.pdf)

I have searched the Internet for source code that actually uses the embedded temperature sensor but have found no example of code that adds either 20 degrees or 25 degrees Celsius to calculate a final result. Apparently no one is using this for ambient temperature readings per se, but the few code examples found use the temperature sensor to correct for drift in the accelerometer / magnetometer of the LSM303D (see [http://www.degruyter.com/downloadpdf/j/afit.2015.36.issue-1/afit-2015-0020/afit-2015-0020.xml](http://www.degruyter.com/downloadpdf/j/afit.2015.36.issue-1/afit-2015-0020/afit-2015-0020.xml).

Changing the sampling rate as suggested in [http://forum.sodaq.com/t/interpreting-payload-data-from-the-sodaqone-universaltracker/374](http://forum.sodaq.com/t/interpreting-payload-data-from-the-sodaqone-universaltracker/374) does not significantly influence the reading (I have experimented with the range 3.125 - 25 Hz). Neither does changing to high resolution (by setting bits M_RES [1:0] of the CTRL5 register to 11).

KKjelstad, 2016-08-12:
@mrjdomingus Adding 4 just ensures that the value is rounded to the correct value when operating with integers.

One degree has eight steps so it means the value 0 to 7 of "int16_t temp" corresponds to 0 to 0.875 degrees. Dividing all values between 0 to 7 by eight using integer results in 0. Which is wrong since a value of 4 (0.5 degrees) should be rounded up to 1.

One improvement I have considered is to use half degrees in the lora message. The current range is -128 to 127 which is wider than the operating temperature range of -40..+85. There is room for a half degree resolution within that range.

alextsam, 2016-08-21:
@mrjdomingus @KKjelstad 
There are some new commits in the repo that address the issue (taking into account the above information).