SodaqMoja/SodaqOne-UniversalTracker #2: OTAA Configuration for MultiTech Conduit

semireg, 2016-08-09:
Hi,

Can you please post the Arduino console or make a video of a successful OTAA connection? I am unable to get a successful OTAA connection using a MultiTech Conduit Gateway. Configuration for multi tech as follows:

```
{
        "lora": {
                "netID": "010203",      /* netID for beacon packets */
                "frequencyBand": "915", /* US="915", EU="868" */
                "frequencySubBand": 7,  /* Sub-band for US operation, 1-8 */
                "rx1DatarateOffset": 0, /* Datarate offset for mote rx window 1 sent in join response (0-3) */
                "rx2Datarate": 12,              /* Datarate for mote rx window 2 sent in join response (7-12) */
                "maxTxPower": 14,       /* Max Tx power (dBm), -6 to 26 */
                "frequencyEU": 867500000 /* center freq for extra EU channels (Hz) */
        },
        "udp": {
                "appPortUp": 1784, /* port for user-developed application use */
                "appPortDown": 1786 /* port for user-developed application use */
        },
        "addressRange": {
                "start": "00:00:00:01", /* address range used for mDots */
                "end": "FF:FF:FF:FE"
        },
        "network": {
                "public": true,    /* set to false for private LoRa network with mDots + Conduit */
                "leasetime": 0,  /* time until mDot join expires (minutes) or 0 for no expiration */
                "eui": "01:01:01:01:01:01:01:01",
                "key": "01:01:01:01:01:01:01:01:01:01:01:01:01:01:01:01"
        },
        "log" : {
                "console" : true,
                "syslog" : false,
                "level" : 30, /* error=10, warn=20, info=30, debug=50, trace=60, max=100 */
                "path": "/var/log/lora-network-server.log"
        },
        "mqtt": {
                "enabled": true
        }
}
```

keestux, 2016-08-10:
May I suggest to post this question on the SODAQ forum, at forum.sodaq.com ?

semireg, 2016-08-10:
You bet, I'll close this out.

closed 2016-08-10 by semireg

semireg, 2016-08-10:
Reference: http://forum.sodaq.com/t/fail-to-connect-to-multitech-conduit-gateway-915-mhz/417