SodaqMoja/SodaqOne-UniversalTracker #15: settings not applied

koos147, 2020-01-31:
I setup the lora tracker and when I am in reach of a good gateweay everything works fine.
however when I am in range of a single channel gateway (Sf12) the board tries to connect twice and then stops.

the board is using SF9 no matter what I change in settings.
and the retry option isn't working.

I plan to place the board on hard to reach locations. I need to make sure they retry at any failure.

```
Settings:

GPS
  GPS (OFF=0 / ON=1)         (gps=): 1
  Fix Interval (min)         (fi=): 15
  Alt. Fix Interval (min)    (afi=): 0
  Alt. Fix From (HH)         (affh=): 0
  Alt. Fix From (MM)         (affm=): 0
  Alt. Fix To (HH)           (afth=): 0
  Alt. Fix To (MM)           (aftm=): 0
  GPS Fix Timeout (sec)      (gft=): 900
  Minimum sat count          (sat=): 3
  Num Coords to Upload       (num=): 1

On-the-move Functionality
  Acceleration% (100% = 8g)  (acc=): 0
  Acceleration Duration      (acd=): 0
  Fix Interval (min)         (acf=): 1
  Timeout (min)              (act=): 10

LoRa
  OTAA Mode (OFF=0 / ON=1)   (otaa=): 1
  Retry conn. (OFF=0 / ON=1) (retry=): 1
  ADR (OFF=0 / ON=1)         (adr=): 1
  ACK (OFF=0 / ON=1)         (ack=): 0
  Spreading Factor           (sf=): 12
  Output Index               (pwr=): 1
  Lora Port                  (lprt=): 1
  DevAddr / DevEUI           (dev=): **********
  AppSKey / AppEUI           (app=): *********
  NWSKey / AppKey            (key=): ********
  Repeat Count               (rep=): 1

Misc
  Cayenne LPP (OFF=0 / ON=1) (cay=): 1
  Temperature Sensor Offset  (temp=): 20
  Status LED (OFF=0 / ON=1)  (led=): 0
  Debug (OFF=0 / ON=1)       (dbg=): 1

```

janvl1234, 2020-02-04:
Hi @koos147 , 

Since you use a single channel gateway you should turn ADR off.
The SF is in the Tracker software for the V1 only set when ADR is off, else it will use the default SF to join. 

The retry is done at the moment it sends, in your case every 15 min is checks if it's connected or it will reconnect.

Best regards,
Jan

koos147, 2020-02-14:
Hi Jan,
thanks you for the reply.

The retry is working as it should be.
it was the message that let me belive the board will wait for a hard reset.
> Failed to reconnect! Hardware-resetting the modem and scheduling a retransmission (if applicable).

i changed the ADR to 0 but the system is still not connecting to a single channel gateway.
the final destenation of the board is in a big city with enough gateways availible. but i need to present them on a small vialge without any lora coverage. so i want to take an cheap gateway with me to make local coverage.

as soon as i am in range of my normale gateway everything works like it should be.

janvl1234, 2020-02-14:
I'm glad to hear that the retry is working as expected.
I have made a note about the debug message.

You still have trouble with the single channel gateway. 
I see you use otaa for the activation, did you try abp?

koos147, 2020-02-28:
Good day, 
with abp the single channel gateway is also working.
now we can start using them in the real world.
Thanks for your support.

closed 2020-02-28 by koos147